orangepi-xunlong/OrangePi_Build #1: install problem

antoniosap, 2017-07-26:
hello, there. i have

....
lib32stdc++6 is already the newest version (6.3.0-12ubuntu2).
lib32stdc++6 set to manually installed.
lib32z1 is already the newest version (1:1.2.11.dfsg-0ubuntu1).
lib32z1-dev is already the newest version (1:1.2.11.dfsg-0ubuntu1).
make is already the newest version (4.1-9.1).
mtools is already the newest version (4.0.18-2ubuntu1).
u-boot-tools is already the newest version (2016.03+dfsg1-6ubuntu2).
bsdtar is already the newest version (3.2.2-2).
pv is already the newest version (1.6.0-1).
debootstrap is already the newest version (1.0.81ubuntu3.2).
qemu-user-static is already the newest version (1:2.8+dfsg-3ubuntu2.3).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 lib32stdc++6-4.8-dbg : Conflicts: lib32stdc++6-4.7-dbg but 4.7.4-3ubuntu12 is to be installed
 lib32stdc++6-4.9-dbg : Conflicts: lib32stdc++6-4.7-dbg but 4.7.4-3ubuntu12 is to be installed
                        Conflicts: lib32stdc++6-4.8-dbg but 4.8.5-4ubuntu4 is to be installed
 lib32stdc++6-5-dbg : Conflicts: lib32stdc++6-4.7-dbg but 4.7.4-3ubuntu12 is to be installed
                      Conflicts: lib32stdc++6-4.8-dbg but 4.8.5-4ubuntu4 is to be installed
                      Conflicts: lib32stdc++6-4.9-dbg but 4.9.4-2ubuntu1 is to be installed
 lib32stdc++6-5-dbg-s390x-cross : Conflicts: lib32stdc++6-4.9-dbg-s390x-cross but 4.9.3-13ubuntu2cross1 is to be installed
 lib32stdc++6-6-dbg : Conflicts: lib32stdc++6-4.7-dbg but 4.7.4-3ubuntu12 is to be installed
                      Conflicts: lib32stdc++6-4.8-dbg but 4.8.5-4ubuntu4 is to be installed
                      Conflicts: lib32stdc++6-4.9-dbg but 4.9.4-2ubuntu1 is to be installed
                      Conflicts: lib32stdc++6-5-dbg but 5.4.1-8ubuntu1 is to be installed
 lib32stdc++6-6-dbg-mips64-cross : Conflicts: lib32stdc++6-5-dbg-mips64-cross but 5.4.1-2ubuntu2cross1 is to be installed
 lib32stdc++6-6-dbg-mips64el-cross : Conflicts: lib32stdc++6-5-dbg-mips64el-cross but 5.4.1-2ubuntu2cross1 is to be installed
 lib32stdc++6-6-dbg-ppc64-cross : Conflicts: lib32stdc++6-5-dbg-ppc64-cross but 5.4.1-2ubuntu2cross1 is to be installed
 lib32stdc++6-6-dbg-s390x-cross : Conflicts: lib32stdc++6-4.9-dbg-s390x-cross but 4.9.3-13ubuntu2cross1 is to be installed
                                  Conflicts: lib32stdc++6-5-dbg-s390x-cross but 5.4.1-8ubuntu1cross1 is to be installed
 lib32stdc++6-6-dbg-sparc64-cross : Conflicts: lib32stdc++6-5-dbg-sparc64-cross but 5.4.1-2ubuntu2cross1 is to be installed
E: Unable to correct problems, you have held broken packages.

JustTheTwoOfUs, 2017-08-24:
gcc-4.9-arm-linux-gnueabihf

antoniosap, 2017-08-26:
i changed script.
the system now create many hundred root tasks and system hangs in 2 minutes

andybeg, 2017-09-20:
solve this problem by remooving from Prepare_toolchain.sh "lib32stdc++" in 14 row and install lib32stdc++ manually

Dantali0n, 2018-01-22:
I had the same problem just make sure that whatever version you change to in "/scripts/Prepare_toolchain.sh" matches your version of gcc. So don't install lib32stdc++-6-dev if you are running gcc 5.4 etc..

antoniosap, 2018-03-03:
solved. a vm and debian 9

closed 2018-03-03 by antoniosap

xuqplus, 2018-05-22:
有帮助 +1

timkoers, 2018-08-02:
@sabur2342 There you go