orangepi-xunlong/OrangePi_Build #2: broken toolchain

leiche5, 2017-08-02:
i downloaded stuff and ran Build_OrangePi.sh which has worked. after changing dir to OrangePiRDA and ran build.sh, script stops because the toolchain folder is empty.

JustTheTwoOfUs, 2017-08-24:
you may just be missing curl, sudo apt-get install curl.
re run the first Build "Build orange pi" and try updating

harnnless, 2017-08-26:
New installation of Ubuntu Server 16.04.3, missing unzip.  sudo apt-get install unzip

warhawk-avg, 2018-07-05:
Enter the directory where the OPi2G build is located...then git clone into the toolchain provided on main build page.
Delete the old toolchain directory
rename the newly cloned directory to directory
That's how I got mine working

orangepi-xunlong, 2018-11-16:
You can refer to #24.

Thanks!

closed 2018-11-16 by orangepi-xunlong