orangepi-xunlong/OrangePi_Build #5: chroot problem

war2973, 2017-09-02:
Hi all,

I cannot build OrangePi Zero Plus2 (H5) image for Ubuntu Xenial:

+ chroot /home/war/OrangePi/OrangePiH5/output/rootfs mount -t proc proc /proc
chroot: failed to run command ‘mount’: Exec format error
+ true
+ chroot /home/war/OrangePi/OrangePiH5/output/rootfs mount -t sysfs sys /sys
chroot: failed to run command ‘mount’: Exec format error
+ true
+ chroot /home/war/OrangePi/OrangePiH5/output/rootfs /second-phase
chroot: failed to run command ‘/second-phase’: Exec format error

any idea?
Thanks

war2973, 2017-09-05:
never mind
I've figured it out.

closed 2017-09-05 by war2973

ewoutp, 2017-09-30:
Same problem here. 
Solved it by creating this file:
```
$ cat /usr/lib/binfmt.d/qemu-aarch64-static.conf
:qemu-aarch64:M::\x7fELF\x02\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\xb7\x00:\xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff:/usr/bin/qemu-aarch64-static:F
```

followed by running:
```
sudo service systemd-binfmt restart
```

Jeepgoing, 2018-06-13:
hello, could you tell me the reason for this question? @war2973 ，thanks.

hougaga, 2019-01-31:
How did you slove it?

ewoutp, 2019-02-05:
Doesn't ring a bell. Sorry

MarkZ2012, 2019-06-15:
sudo apt purge qemu-user-static
sudo apt install qemu-user-static

This works OK for me on orange pi plus.