orangepi-xunlong/OrangePi_Build #6: cannot create ... test-dev-null

andybeg, 2017-09-20:
build ubuntu xenial process ended with error

=======================================================
Creating Ubuntu/Debian SD Card instalation for OrangePI
=======================================================

Creating root filesistem in local directory /media/barsuk/ext4.720/OrangePiH3/output/linux-xenial...

DEBOOTSTRAP, FIRST STAGE
this
please wait \/usr/sbin/debootstrap: 1368: /usr/sbin/debootstrap: cannot create /media/barsuk/ext4.720/OrangePiH3/output/linux-xenial/test-dev-null: Permission denied
E: Cannot install into target '/media/barsuk/ext4.720/OrangePiH3/output/linux-xenial' mounted with noexec or nodev
OK.                        
Creating "sources.list"
Error while loading /bin/bash: No such file or directory
================================================
ERROR on second-stage, instalation NOT FINISHED.
================================================

andybeg, 2017-09-21:
frash install solved problem

zhangximin, 2018-04-04:
After google, some says encrypted directory cause this in home path.I clone into other path resolved the issue.

closed 2018-11-16 by orangepi-xunlong

YunusWang, 2019-07-12:
I also had the same problem. As follow

Creating root filesistem in local directory linux-xenial...

DEBOOTSTRAP, FIRST STAGE
OK.                        
Creating "sources.list"
Error while loading /bin/bash: No such file or directory
================================================
ERROR on second-stage, instalation NOT FINISHED.
================================================

So, is there a solution ?

KF5JWC, 2022-02-13:
Ran into this type of issue from google.

```
P: Running debootstrap... 
/usr/share/debootstrap/functions: line 1578: /vyos/build/chroot/test-dev-null: Permission denied
E: Cannot install into target '/vyos/build/chroot' mounted with noexec or nodev
```

My issue was also because of ecryptfs home directory.

```
$ mount | grep home | grep Private
/home/.ecryptfs/user/.Private on /home/user type ecryptfs (rw,nosuid,nodev,relatime,ecryptfs_fnek_sig=...
```

In short, use another directory for this task. I have root on my box, so I used `/opt`, but if needed you may be able to mount a `tmpfs` that you build in.