orangepi-xunlong/OrangePi_Build #9: downloading issue: no packet lib32z1 lib32z1-dev

qdsjx1989, 2017-11-20:
when I downloading the kernel file:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package lib32z1
E: Unable to locate package lib32z1-dev

JYKeith123, 2018-03-20:
I found this libraries are not for ARM processor. - [link](https://github.com/nylas/sync-engine/issues/414)
So I commentize it from [this](https://github.com/orangepi-xunlong/OrangePi_Build/blob/master/lib/PrepareTool.sh).
But still there is a problem. I'm building os for i96 (RDA or 2G IOT), but 
they have been using H3 toolchain I don't know why.

https://github.com/orangepi-xunlong/OrangePi_Build/blob/master/lib/i96SDK_BuildEnvironment.sh
https://github.com/orangepi-xunlong/OrangePi_Build/blob/master/lib/RDASDK_BuildEnvironment.sh
both use below toolchain.
`toolchain="https://codeload.github.com/orangepi-xunlong/OrangePiH3_toolchain/zip/master"`

I tried to build my own linux with my orangepi i96, and of course, it failed because the toolchain for H3 can not be executed on i96.

orangepi-xunlong, 2018-11-16:
It is recommended to use the SDK on Ubuntu 14.04.

and i96 and h3 have the same toolchain.

Thanks!

closed 2018-11-16 by orangepi-xunlong

RakibFiha, 2020-06-16:
I am having this same issue.