orangepi-xunlong/OrangePi_Build #12: 生成的winplus镜像启动不了

dksslq, 2018-02-04:
将生成的镜像刷入tf卡之后无法启动，而且第二个分区，也就是根分区有超级块错误，如何解决？

dksslq, 2018-11-18:
没看到log文件，但是发现问题所在了，启动时观察uart调试信息时发现boot0成功启动，但是转交给uboot时显示uboot size为0,开机过程在这里停止了

问题脚本：OrangePiA64_scripts/uboot_compile.sh
73-78:
# Merge u-boot.bin infile outfile mode [secmonitor | secos | scp]
$MERGE_TOOLS/merge_uboot  u-boot.bin  bl31.bin  u-boot-merged.bin secmonitor
$MERGE_TOOLS/merge_uboot  u-boot-merged.bin  scp.bin  u-boot-merged2.bin scp

# Merge uboot and dtb
$MERGE_TOOLS/update_uboot_fdt u-boot-merged2.bin orangepi.dtb u-boot-with-dtb.bin

应该merge工具的问题，换成armbian源码中的boot0img工具就可以启动了:
# 融合dtb到uboot
$UBT_SRC/tools/boot0img -s $PREBUILT/scp.bin -d $PREBUILT/bl31.bin -u $UBOOT_ORIG -e -F $UBT_OUT/$DT_FILE -o $UBT_OUT/u-boot-with-dtb.bin

最后附上自己的opiwin构建系统:
https://github.com/dksslq/debian-opiwin

另外 opiwin的千兆网卡是有严重的丢包缺陷的，请问其他型号的板子有这样的缺陷码？比如zero plus

orangepi-xunlong, 2020-03-03:
已修复

closed 2020-03-03 by orangepi-xunlong