orangepi-xunlong/OrangePi_Build #18: Step 2 not working !!

vincegre, 2018-05-01:
Hi

As you don't supply an image with at least basic drivers for Pi One Plus, I tried to compile it myself but at step 2 when I run the Build_OrangePi.sh script it asks me for my root password and that's it, it returns to prompt without any errors but doing nothing !!
Fedora 27 (host system on which I try to build the image !)

Thanks

Vincèn

EvilOlaf, 2018-05-03:
Add your user account to sudoers and provided your users password when asked.

vincegre, 2018-05-03:
> Add your user account to sudoers and provided your users password when asked.

Sorry but my account is already in sudoers and I supplied the correct password !  Any other ideas ? trying to compile on Fedora is not going to be a problem ? on I should do that from an Ubuntu host ?

EvilOlaf, 2018-05-03:
As I successfully ran the scripts successfully on Ubuntu server 16.04 amd64 (except a small issue for which I provided a fix https://github.com/orangepi-xunlong/OrangePiH6_scripts/pull/1) I'd recommend yes. Retry using an Ubuntu (based) OS.

vincegre, 2018-05-29:
@EvilOlaf thanks but I gave up considering how manufacturer itself don't care at all about it :(

closed 2018-05-29 by vincegre

EvilOlaf, 2018-05-29:
That's the price for low cost hardware I guess g ¯\_(ツ)_/¯

vincegre, 2018-05-29:
> That's the price for low cost hardware I guess ¯\(ツ)/¯
Well you can do some cheap hardware and still good software (but well chinese companies are most of time famous for their so poor softwares/firmwares unless hardware is really nice sometimes.....