orangepi-xunlong/OrangePi_Build #19: 编译 OrangePiH3 时报错 /second-stage: line 232: /bin/sleep: cannot execute binary file: Exec format err

RedKeys, 2018-05-03:
Creating "sources.list"
DEBOOTSTRAP, SECOND-STAGE
/second-stage: line 232: /bin/sleep: cannot execute binary file: Exec format err                             
please wait \/second-stage: line 206: /bin/sleep: cannot execute binary file: Exec format error
please wait |/second-stage: line 206: /bin/sleep: cannot execute binary file: Exec format error
please wait //second-stage: line 206: /bin/sleep: cannot execute binary file: Exec format error
please wait -/second-stage: line 206: /bin/sleep: cannot execute binary file: Exec format error
please wait \/second-stage: line 206: /bin/sleep: cannot execute binary file: Exec format error
please wait |/second-stage: line 206: /bin/sleep: cannot execute binary file: Exec format error
please wait //second-stage: line 206: /bin/sleep: cannot execute binary file: Exec format error
please wait -/second-stage: line 206: /bin/sleep: cannot execute binary file: Exec format error
please wait \/debootstrap/debootstrap: line 382: /bin/cat: cannot execute binary file: Exec format error
/second-stage: line 206: /bin/sleep: cannot execute binary file: Exec format error
ERROR.
================================================
ERROR on second-stage, instalation NOT FINISHED.
================================================

EvilOlaf, 2018-05-03:
Interesting. Pure guess but I think the script does not properly chroot into the new environment and trying to execute an arm binary on x86/amd64 which obviously does not work.

warhawk-avg, 2018-07-05:
Should this be run as a chroot or in a docker under armhf?

frol, 2018-08-02:
I have bumped into this issue when I used Docker container to build the rootfs. `--privileged` flag to the `docker run` command fixed the issue.

Just for the record, here is how I configure the build environment:

```
$ docker run -it --rm "$(pwd):/mnt" --privileged ubuntu:18.04 bash
root@xxx:/# apt update
root@xxx:/# apt install git make whiptail sudo curl unzip figlet device-tree-compiler
root@xxx:/# echo Europe/Kiev > /etc/timezone 
```

Then I can follow the usual path of `./Build_OrangePi.sh && cd ../OrangePiH3 && ./build.sh`.

suniltop, 2018-08-16:
Same issue here for orange-pi-zero.

Also I have made a test manually but no solution 
In fact bash screen has been opened but every command returns same error that exec format error
Command is below

sudo chroot /home/rt/opizero/OrangePiH2/output/linux-jessie qemu-arm-static -cpu cortex-a9 /bin/bash

Any suggestion ?

ouweipeng8888, 2019-01-12:
echo ":arm:M::\x7fELF\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x28\x00:\xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff:/usr/bin/qemu-arm-static:"> /proc/sys/fs/binfmt_misc/register

orangepi-xunlong, 2019-12-06:
We have updated the compilation system of H3, welcome to use it.
Supported build environment is Ubuntu Bionic 18.04 x64.

closed 2019-12-06 by orangepi-xunlong