orangepi-xunlong/OrangePi_Build #22: compile orangepi zero plus2 image，boot cmd error

zhen8838, 2018-05-28:
I use ./build.sh build the OrangePiH5_Zero_Plus2.img , when i run in borad , can't boot success like this:

```sh
sunxi#boot
Loading orangepi uEnv.txt from 40000000 ...
reading uEnv.txt
** Unable to read file uEnv.txt **
reading boot.scr
** Unable to read file boot.scr **
Booting with defaults ...
Loading orangepi orangepi/OrangePiH5.dtb from 44000000 ...
reading orangepi/OrangePiH5.dtb
** Unable to read file orangepi/OrangePiH5.dtb **
Loading orangepi orangepi/uImage from 4007ffc0 ...
reading orangepi/uImage
** Unable to read file orangepi/uImage **
```
I check the sd crad /BOOT ,nothing.and compile the u-boot can't build the boot.scr .

zhen8838, 2018-05-28:
now i cpoy the OrangePiH5.dtb uEnv.txt uImage to /BOOT,but the boot.scr i use the nanopi-neo-plus2 build. i am not sure it can be success.
the new error is :
```sh
sunxi#boot
Loading orangepi uEnv.txt from 40000000 ...
reading uEnv.txt
[mmc]: blkcnt should not be 0
114 bytes read in 2 ms (55.7 KiB/s)
Loading boot environment ...
reading boot.scr
1185 bytes read in 2 ms (578.1 KiB/s)
Booting with script ...
Unknown command 'source' - try 'help'
```
now what can i do?

AndreyDatakam, 2019-03-07:
Hello) If you solved that problem, can you tell how did you do that. Thanks.

zhen8838, 2019-03-08:
@AndreyDatakam  too long ago, I already forget. but I refer you use Armbian

orangepi-xunlong, 2020-02-27:
Hi, the problem has been solved in the latest code.

closed 2020-02-27 by orangepi-xunlong