orangepi-xunlong/OrangePi_Build #23: OPi Win+ Toolchain issue

silentbogo, 2018-05-28:
Trying to build a distro for OrangePi Win Plus...
I'm getting an error, regarding unpacking of the toolchain archive (toolchain.tar.gz)
First error was due to the wrong location of the archive in the build script (OrangePiA64/toolchain/ instead of OrangePiA64/toolchain/toolchain_tar/)
After moving the tarball to that location I've got this error:
[image]

Only a moment later I noticed that the file size was only zero bytes....

EvilOlaf, 2018-05-29:
Probably known issue.
Make sure `curl` is installed.

silentbogo, 2018-05-29:
Yep, curl is installed. All other components get downloaded no problem. 
Afterwards (after manually creating folder and copying toolchain), everything seems to work and compile just fine.
I'll look into scripts later tonight and try to figure out why the toolchain isn't downloading.

warhawk-avg, 2018-07-05:
Enter the directory where the OPi2G build is located...then git clone into the toolchain provided on main build page.
Delete the old toolchain directory
rename the newly cloned directory to directory
That's how I got mine working

orangepi-xunlong, 2020-03-03:
Has been updated.

closed 2020-03-03 by orangepi-xunlong