orangepi-xunlong/OrangePi_Build #24: Fails to install toolchain for H5

ril3y, 2018-05-30:
After running ./build.sh
```
/home/ril3y/dev/OrangePiH5
 Uncompress toolchain..
cat: '/home/ril3y/dev/OrangePiH5/toolchain/toolchain_tar/toolchain*': No such file or directory
```

Not sure what the error is.

ril3y, 2018-06-05:
nothing?

Jeepgoing, 2018-06-13:
make sure `curl` is installed.

warhawk-avg, 2018-07-06:
I had to manually git clone the toolchain into the directory..then move/rename the OrangePiH5_toolchain to just plain toolchain to get it to compile
cd /home/ril3y/dev/OrangePiH5
git clone https://github.com/orangepi-xunlong/OrangePiH5_toolchain.git
sudo rm -rf toolchain
sudo mv OrangePiH5_toolchain toolchain

Make sure you have run the OrangePi_Build/lib/PrepareTool.sh to install the required dependencies for the build

abo-github, 2018-08-11:
Okay ... that's wired - but now it works!

I really had to git clone the repo **OrangePi_Build** with
`
git clone https://github.com/orangepi-xunlong/OrangePi_Build.git
`
Downloading the zip file and extracting it didn't work for me.

So you have to follow exactly the instructions on
[https://github.com/orangepi-xunlong/OrangePi_Build](url)

orangepi-xunlong, 2018-11-16:
The lack of access to the toolchain is due to a network problem that caused the download to fail. You may be able to try to re-download the code through the OrangePi_Build. A separate download for OrangePiH5_toolchain.git is not recommended. 

Thanks!

closed 2018-11-16 by orangepi-xunlong