orangepi-xunlong/OrangePi_Build #26: Fails to install toolchain for i96 and 2G-IOT

boulbaal, 2018-06-28:
Fails to build # OrangePi 2G-IOT        RDA8810      # OrangePi i96           RDA8810

the bash script "i96SDK_BuildEnvironment.sh and RDASDK_BuildEnvironment.sh" refers to a wrong toolchain.-->toolchain="https://codeload.github.com/orangepi-xunlong/OrangePiH3_toolchain/zip/master"
it is possible to provide the correct toochain for the RDA8810pl.

warhawk-avg, 2018-07-05:
Enter the directory where the OPi2G build is located...then git clone into the toolchain provided on main build page.
Delete the old toolchain directory
rename the newly cloned directory to directory
That's how I got mine working

orangepi-xunlong, 2018-11-16:
The toolchain of 2G-IOT and H3 are the same.

thanks!

closed 2018-11-16 by orangepi-xunlong