orangepi-xunlong/OrangePi_Build #30: /bin/sleep

webstersx01, 2018-07-23:
Building on ubuntu 18.04 x64, this error: please wait \/second-stage: line 206: /bin/sleep: cannot execute binary file: Exec format error

sabur2342, 2018-07-27:
Programs that must be installed before the assembly of OS^
apt-get install curl libcurl4 libcurl4-openssl-dev
apt-get install lib32stdc++6
apt-get install git fakeroot build-essential ncurses-dev xz-utils libssl-dev bc
apt-get install linux-source
apt-get install debootstrap
apt-get install schroot
apt-get install binfmt-support qemu qemu-user-static
apt-get install lib32ncurses5 lib32ncurses5-dev libusb-1.0-0-dev

webstersx01, 2018-07-28:
Okay, now I can compile ubuntu/debian, but on raspbian compilation I have another trouble: Error while loading /bin/bash: No such file or directory

webstersx01, 2018-07-31:
Thank you very much!

closed 2018-07-31 by webstersx01