orangepi-xunlong/OrangePi_Build #31: Compilation freezes on Debian 9.4.0

timkoers, 2018-08-01:
I am trying to compile for the Orange Pi Zero and on both my VM and my secondary laptop (which hasn't got the best specs) the operating system freezes without any errors. It keeps on reading from the disk though, as if it was reading files to compile. How do I fix this?

sabur2342, 2018-08-01:
Attach screenshots or describe the problem in more detail.

timkoers, 2018-08-01:
I'm afraid that is all I know. It started compiling the kernel and the whole system (my laptop and VM) freezes, I am not compiling it at the Orange Pi itself. It doesn't stop at a certain file and it looks like the processor is being used to much.

sabur2342, 2018-08-01:
I recommend checking:
1. How many cores and RAM are allocated to the virtual machine;
2. Install ubuntu 18.04 on the VM and try on it;
3. I used this guide - [How to build the images](https://github.com/skycoin/Orange-Pi-H5/wiki/How-to-build-the-images)

But the image created by the script recorded on microsd was not loaded. More precisely, the eth-adapter was not started. I had to take the already created .img-image of the OS from the website [OS for OrangePi](http://www.orangepi.org/downloadresources) and replace the partition of the /boot and /roofs replaced with the data created by the script, then everything worked.

Programs that must be installed before the assembly of OS^
apt-get install curl libcurl4 libcurl4-openssl-dev
apt-get install lib32stdc++6
apt-get install git fakeroot build-essential ncurses-dev xz-utils libssl-dev bc
apt-get install linux-source
apt-get install debootstrap
apt-get install schroot
apt-get install binfmt-support qemu qemu-user-static
apt-get install lib32ncurses5 lib32ncurses5-dev libusb-1.0-0-dev

sabur2342, 2018-08-01:
Something else. On Debian 9.4 to assemble the system did not work out. It turned out on Debian 9.5 and ubuntu 18.04.  Make sure to: apt update and apt upgrade.

timkoers, 2018-08-01:
There are 4 cores and 4GB of ram allocated to the VM.
I think my laptop has the same specs, I've not checked that yet since I got it as a workbench laptop 2 days ago.

The build script installs all those programs right?
Again, the compilation is not failing, the operating system the source code is being compiled on freezes during the first 2 minutes of the compilation

timkoers, 2018-08-01:
Aah thanks!

sabur2342, 2018-08-01:
The guidance How to build the images is correct. 
If there is an error: Script_to_dts: Can not find [sun50iw2p1-cheetah-p1.dts]
Locate the file: ".sun50iw2p1-soc.dtb.dts" and copy it to the directory specified by the script when displaying the error message.

timkoers, 2018-08-01:
[image]
It froze once again on my VM running Debian 9.5. The disk indicator at the bottom shows it is still reading or writing to the disk as it keeps blinking red and green.
I'll try Ubuntu now

timkoers, 2018-08-01:
I've got another error now:

[image]

sabur2342, 2018-08-01:
I used only stable assemblies of Debian and Ubuntu. Update the operating system Ubuntu. You on the screenshot indicate that the system has available updates. Does the display of the script work stop or something else happen on the Debian?

sabur2342, 2018-08-02:
PrepareTool.sh script in walks in OrangePi_Build. These files are needed to run the scripts to load the source code for your OrangePi. I did the assembly for OrangePi Prime and everything turned out. I'll try to assemble for OrangePi Zero. Let's see what happens.

sabur2342, 2018-08-04:
Unfortunately it was not possible to collect from source codes Orange Pi Zero.
In Ubunt immediately gives an error after the start ./build.sh
[image]

After the launch build uboot. Did not automatically download the toolchain.
OrangePi Zero plus build on Allwinner H2+ Soc, the offfically maintained repository as follows:
I had to download manually: git clone https://github.com/orangepi-xunlong/OrangePiH3_toolchain.git
[image]

timkoers, 2018-08-04:
I've done that and I've removed the conflicting package from the script. It now builds successfully, but it doesn't generate and .IMG file

[image]<https://user-images.githubusercontent.com/41253830/43673592-69e6e948-97ce-11e8-885e-5a7b93940100.png>

[image]<https://user-images.githubusercontent.com/41253830/43673595-8c0c248e-97ce-11e8-91b7-9e569d9a0f38.png>

sabur2342, 2018-08-04:
If you do not create an image, but there are files for the kernel and root. That can be manually copied to the OS image for [OrangePi Zero](http://www.orangepi.org/downloadresources/).

alex-eri, 2018-08-11:
Looks like fork-bomb

timkoers, 2018-08-12:
@alex-eri I've changed that in a pull request of the script repo. That request hasn't been accepted yet, it would help if you drew some attention too to that [pull request](https://github.com/orangepi-xunlong/OrangePiH2_scripts/pull/2)

afilis, 2018-08-23:
@sabur2342 I had the same problem. I fixed it by replacing lib32stdc++ to lib32stdc++6 in scripts/Prepare_toolchain.sh.