orangepi-xunlong/OrangePi_Build #34: Does build toolkit only support i386 platform?

qxj, 2018-08-17:
I tried to run `build.sh` to build release image for OrangePi Prime in Debian9 x64, but always failed at chroot. 

Logs here, plz help:

```
Extracting ... + bsdtar -xpf /home/jqian/works/OrangePiH5/external/ArchLinuxARM-aarch64-latest.tar.gz -C /home/jqian/works/OrangePiH5/output/rootfs
+ echo 'OK'
OK
+ cp /usr/bin/qemu-aarch64-static /home/jqian/works/OrangePiH5/output/rootfs/usr/bin
+ cp /usr/bin/qemu-arm-static /home/jqian/works/OrangePiH5/output/rootfs/usr/bin
+ cat
+ chmod a+x /home/jqian/works/OrangePiH5/output/rootfs/usr/sbin/policy-rc.d
+ case $DISTRO in
+ mv /home/jqian/works/OrangePiH5/output/rootfs/etc/resolv.conf /home/jqian/works/OrangePiH5/output$rootfs/etc/resolv.conf.dist
+ cp /etc/resolv.conf /home/jqian/works/OrangePiH5/output/rootfs/etc/resolv.conf
+ sed -i 's|CheckSpace|#CheckSpace|' /home/jqian/works/OrangePiH5/output/rootfs/etc/pacman.conf
+ do_chroot pacman -Rsn --noconfirm linux-aarch64
+ cmd='pacman -Rsn --noconfirm linux-aarch64'
+ chroot /home/jqian/works/OrangePiH5/output/rootfs mount -t proc proc /proc
chroot: failed to run command 'mount': Exec format error
+ true
+ chroot /home/jqian/works/OrangePiH5/output/rootfs mount -t sysfs sys /sys
chroot: failed to run command 'mount': Exec format error
+ true
+ chroot /home/jqian/works/OrangePiH5/output/rootfs pacman -Rsn --noconfirm linux-aarch64
chroot: failed to run command 'pacman': Exec format error
+ chroot /home/jqian/works/OrangePiH5/output/rootfs umount /sys
chroot: failed to run command 'umount': Exec format error
+ chroot /home/jqian/works/OrangePiH5/output/rootfs umount /proc
chroot: failed to run command 'umount': Exec format error
```

EvilOlaf, 2018-08-17:
Retry with Ubuntu 16.04 x64 as OS

qxj, 2018-08-19:
I tried with ubuntu 16.04 x64, the same error.

```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.5 LTS"
```

qxj, 2018-08-19:
It looks like the `do_chroot` function contains bugs in the script [00_rootfs_build.sh](https://github.com/orangepi-xunlong/OrangePiH5_scripts/blob/5d466cbf2e33d4914d4bca035f7a67e7c4ae1b2a/00_rootfs_build.sh#L151). `qemu-aarch64-static` is missing!

The correct `do_chroot` should be this:

```bash
do_chroot() {
    cmd="$@"
    chroot "$DEST" qemu-aarch64-static /usr/bin/mount -t proc proc /proc || true
    chroot "$DEST" qemu-aarch64-static /usr/bin/mount -t sysfs sys /sys || true
    chroot "$DEST" qemu-aarch64-static $cmd
    chroot "$DEST" qemu-aarch64-static /usr/bin/umount /sys
    chroot "$DEST" qemu-aarch64-static /usr/bin/umount /proc
}
```

parrotgeek1, 2018-10-11:
You shouldn't need this if you have binfmt_misc set up for qemu

actsasflinn, 2019-03-09:
This seems to resolve with https://github.com/orangepi-xunlong/OrangePi_Build/issues/5#issuecomment-333330327

tuxd3v, 2019-03-21:
I think the correct answer will be ...
Install in the HOST the qemu user package:
apt-get install qemu-user-static

Because even solving that,
if you are cross-debootstrapping, it will fail in second bootstrap stage...

yoojia, 2019-04-18:
> I think the correct answer will be ...
> Install in the HOST the qemu user package:
> apt-get install qemu-user-static
> 
> Because even solving that,
> if you are cross-debootstrapping, it will fail in second bootstrap stage...

You are RIGHT, works for me: 
OS: Ubuntu 18.04
BOARD: OrangePi 3

Try reinstall **qemu-user-static**
```
sudo apt purge qemu-user-static
sudo apt install qemu-user-static
```

celebum, 2019-04-23:
sudo mount -i -o remount,exec,dev /home/jqian/works

cronnosli, 2019-05-26:
You can try 
```bash
apt-get install qemu-utils qemu-efi-aarch64 qemu-system-arm
```
It works for me.

makito013, 2020-03-12:
> You can try
> 
> ```shell
> apt-get install qemu-utils qemu-efi-aarch64 qemu-system-arm
> ```
> 
> It works for me.

THANKS

closed 2020-03-15 by qxj