orangepi-xunlong/OrangePi_Build #39: build OrangePi zero uboot Error

fcsw2fcsw2, 2018-12-05:
in OrangePiH2 directory 
./build.sh
Compile uboot for OrangePi zero
but 
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/fcsw2fcsw2/Desktop/OrangePi/OrangePi_zero/OrangePiH2/uboot/sunxi_spl/fes_init'
Makefile:515: recipe for target 'fes' failed
make: *** [fes] Error 2
make: *** Waiting for unfinished jobs....
make[1]: *** No rule to make target '/home/fcsw2fcsw2/Desktop/OrangePi/OrangePi_zero/OrangePiH2/uboot/sunxi_spl/boot0/main/boot0.lds', needed by '/home/fcsw2fcsw2/Desktop/OrangePi/OrangePi_zero/OrangePiH2/uboot/sunxi_spl/boot0/boot0.lds'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/fcsw2fcsw2/Desktop/OrangePi/OrangePi_zero/OrangePiH2/uboot/sunxi_spl/boot0'
Makefile:520: recipe for target 'boot0' failed
make: *** [boot0] Error 2
make[1]: *** No rule to make target '/home/fcsw2fcsw2/Desktop/OrangePi/OrangePi_zero/OrangePiH2/uboot/sunxi_spl/sbrom/main/sboot.lds', needed by '/home/fcsw2fcsw2/Desktop/OrangePi/OrangePi_zero/OrangePiH2/uboot/sunxi_spl/sbrom/sboot.lds'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/fcsw2fcsw2/Desktop/OrangePi/OrangePi_zero/OrangePiH2/uboot/sunxi_spl/sbrom'
Makefile:546: recipe for target 'sboot' failed
make: *** [sboot] Error 2

PC:Ubuntu16.04 X64

robertsmedshammar, 2019-03-07:
I get the same error, anyone knowns if this has been resolved?

tatarysh, 2019-03-24:
same error

driver003, 2019-05-30:
I found this solution.Copy this [here](https://github.com/friendlyarm/h3_lichee/tree/master/brandy/u-boot-2011.09/sunxi_spl) directory lds files to the H2 directory ,it is work.

tatarysh, 2019-05-30:
@driver003 thanks, I'll check it :)

orangepi-xunlong, 2019-12-06:
We have updated the compilation system of H2+, welcome to use it. 
Supported build environment is Ubuntu Bionic 18.04 x64.

closed 2019-12-06 by orangepi-xunlong