orangepi-xunlong/OrangePi_Build #43: Not able install Docker after Ubuntu Server installation on Orange Pi 3

UniverseXXX, 2019-02-05:
I've got the following error when trying to install docker on ubuntu server on Orange Pi 3:
```
Setting up docker-ce (5:18.09.1~3-0~ubuntu-xenial) ...
update-alternatives: using /usr/bin/dockerd-ce to provide /usr/bin/dockerd (dockerd) in auto mode
Job for docker.service failed because the control process exited with error code. See "systemctl status docker.service" and "journalctl -xe" for details.
invoke-rc.d: initscript docker, action "start" failed.
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2019-02-05 19:11:51 UTC; 13ms ago
     Docs: https://docs.docker.com
  Process: 5874 ExecStart=/usr/bin/dockerd -H fd:// (code=exited, status=1/FAILURE)
 Main PID: 5874 (code=exited, status=1/FAILURE)

Feb 05 19:11:51 OrangePi systemd[1]: Failed to start Docker Application Container Engine.
Feb 05 19:11:51 OrangePi systemd[1]: docker.service: Unit entered failed state.
Feb 05 19:11:51 OrangePi systemd[1]: docker.service: Failed with result 'exit-code'.
dpkg: error processing package docker-ce (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for systemd (229-4ubuntu21.15) ...
Processing triggers for libc-bin (2.23-0ubuntu10) ...
Errors were encountered while processing:
 docker-ce
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

kozobrod, 2019-02-13:
have the same problem, please, include aufs in kernel

obenturk, 2019-03-03:
Same issue overlay/overlay2, aufs needed.
Any comment how to add these in kernel?

serhiiz, 2019-04-12:
There's now [orangepi-xunlong\OrangePiH6_Linux4_9](https://github.com/orangepi-xunlong/OrangePiH6_Linux4_9) repository and I see some mentions of overlayfs: [fs\overlayfs](https://github.com/orangepi-xunlong/OrangePiH6_Linux4_9/tree/master/fs/overlayfs). Would it mean it's coming?

Meanwhile, here's the docker daemon's (5:18.09.5~3-0~ubuntu-xenial) output:
```
root@OrangePi:~# uname -a
Linux OrangePi 4.9.118+ #5 SMP PREEMPT Mon Mar 4 18:30:41 CST 2019 aarch64 aarch64 aarch64 GNU/Linux
root@OrangePi:~# /usr/bin/dockerd
INFO[2019-04-12T04:12:08.273355335Z] parsed scheme: "unix"                         module=grpc
INFO[2019-04-12T04:12:08.273738377Z] scheme "unix" not registered, fallback to default scheme  module=grpc
INFO[2019-04-12T04:12:08.274054293Z] parsed scheme: "unix"                         module=grpc
INFO[2019-04-12T04:12:08.274140876Z] scheme "unix" not registered, fallback to default scheme  module=grpc
ERRO[2019-04-12T04:12:08.284837362Z] 'overlay' not found as a supported filesystem on this host. Please ensure kernel is new enough and has overlay support loaded.  storage-driver=overlay2
ERRO[2019-04-12T04:12:08.295090807Z] AUFS was not found in /proc/filesystems       storage-driver=aufs
INFO[2019-04-12T04:12:08.306651501Z] ccResolverWrapper: sending new addresses to cc: [{unix:///run/containerd/containerd.sock 0  <nil>}]  module=grpc
INFO[2019-04-12T04:12:08.306807417Z] ClientConn switching balancer to "pick_first"  module=grpc
INFO[2019-04-12T04:12:08.307087584Z] pickfirstBalancer: HandleSubConnStateChange: 0x44209d85f0, CONNECTING  module=grpc
INFO[2019-04-12T04:12:08.308179290Z] pickfirstBalancer: HandleSubConnStateChange: 0x44209d85f0, READY  module=grpc
INFO[2019-04-12T04:12:08.308499165Z] ccResolverWrapper: sending new addresses to cc: [{unix:///run/containerd/containerd.sock 0  <nil>}]  module=grpc
INFO[2019-04-12T04:12:08.308607082Z] ClientConn switching balancer to "pick_first"  module=grpc
INFO[2019-04-12T04:12:08.309040623Z] pickfirstBalancer: HandleSubConnStateChange: 0x44209d8880, CONNECTING  module=grpc
INFO[2019-04-12T04:12:08.310056163Z] pickfirstBalancer: HandleSubConnStateChange: 0x44209d8880, READY  module=grpc
ERRO[2019-04-12T04:12:08.320997232Z] 'overlay' not found as a supported filesystem on this host. Please ensure kernel is new enough and has overlay support loaded.  storage-driver=overlay
ERRO[2019-04-12T04:12:08.321261649Z] Failed to built-in GetDriver graph devicemapper /var/lib/docker
INFO[2019-04-12T04:12:08.329143472Z] Graph migration to content-addressability took 0.00 seconds
WARN[2019-04-12T04:12:08.330129179Z] Your kernel does not support cgroup cfs period
WARN[2019-04-12T04:12:08.330272137Z] Your kernel does not support cgroup cfs quotas
WARN[2019-04-12T04:12:08.330382762Z] Unable to find blkio cgroup in mounts
WARN[2019-04-12T04:12:08.330441470Z] Unable to find cpuset cgroup in mounts
WARN[2019-04-12T04:12:08.330728970Z] mountpoint for pids not found
Error starting daemon: Devices cgroup isn't mounted
```

SuYanw, 2019-06-05:
same problem here

AvMu, 2019-06-08:
same problem Ubuntu 16.04.6 LTS
coz docker wont install on 18.04.2 ( E: could not locate package )

baiywt, 2019-06-17:
The following script from the Docker project will check various Kernel features to see whether any changes need to be made to support Docker and Swarm Mode fully.
https://raw.githubusercontent.com/docker/docker/master/contrib/check-config.sh
./check-config.sh kernel/.config

orangepi-xunlong, 2020-02-26:
[Linux4.9](https://github.com/orangepi-xunlong/OrangePiH6_Linux4_9/commit/4d8edacbbecf886cb68faa3c7f00076af90c2ed0) has added a docker-related kernel configuration
[OrangePi_3_ubuntu_xenial_desktop_linux4.9.118_v2.0.3.tar.gz](http://www.orangepi.org/downloadresources/) is the latest images.

closed 2020-02-26 by orangepi-xunlong