orangepi-xunlong/OrangePi_Build #44: Unable to run OrangePi_Build.sh on Orange Pi 3

paszkoski, 2019-02-08:
Hello,
I've got a problem with OrangePi_Build.sh script on Orange Pi 3 with Debian Jessie installed. 
When I run it and type root password I get error:
`Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package lib32z1
E: Unable to locate package lib32z1-dev`

I've read somewhere that those packages are not available for ARM architecture - is this true? And if so, are they really needed on Orange Pi 3? 

Best regards,
Michal

AndreiCherniaev, 2021-04-12:
The same problem while I try to build Linux on my ARM. 
It looks like @orangepi-xunlong support x86_64 only for Linux building. 
Does @orangepi-xunlong plan to allow build Linux on Orange Pi?

orangepi-xunlong, 2021-04-13:
Yes, it only supports Ubuntu18.04 X86_64, no other platforms. There are no plans to support other platforms.

closed 2021-04-13 by orangepi-xunlong