orangepi-xunlong/OrangePi_Build #51: UBoot build error Kernel 4.9

Dreddits, 2019-03-07:
Uboot does not output files from running build.sh script for version 4.9, will update with error code and missing files. think it's  a missing libary in Linaro part of toolchian compile.

Dreddits, 2019-03-07:
Compile U-boot......
Configuring for sun50iw6p1 - Board: sun50iw6p1, Options: SUN50IW6P1
/home/dredds/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
/home/dredds/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
dirname: missing operand
Try 'dirname --help' for more information.
  GEN     include/autoconf.mk.dep
/home/dredds/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
  GEN     include/autoconf.mk
/home/dredds/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  HOSTCC  scripts/basic/fixdep
  UPD     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
/home/dredds/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
  UPD     include/generated/version_autogenerated.h
  CC      lib/asm-offsets.s
  CC      arch/arm/lib/asm-offsets.s
/home/dredds/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
/home/dredds/OrangePiH6/uboot/./Kbuild:34: recipe for target 'lib/asm-offsets.s' failed
make[1]: *** [lib/asm-offsets.s] Error 127
make[1]: *** Waiting for unfinished jobs....
/home/dredds/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
/home/dredds/OrangePiH6/uboot/./Kbuild:83: recipe for target 'arch/arm/lib/asm-offsets.s' failed
make[1]: *** [arch/arm/lib/asm-offsets.s] Error 127
Makefile:1134: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2

leeboby, 2019-03-08:
Please install lib32stdc++6

$ sudo apt-get install lib32stdc++6

closed 2019-03-08 by orangepi-xunlong