orangepi-xunlong/OrangePi_Build #52: Error Orangepi_Oneplus kernel 4.9

tuxd3v, 2019-03-07:
When I go to compile kernel and uboot,
Kernel Compiles OK, but uboot fails with message:
"
H6/OrangePiH6$ ./build.sh 
/home/tuxd3v/Desktop/H6/OrangePiH6

Compile U-boot......
Configuring for sun50iw6p1 - Board: sun50iw6p1, Options: SUN50IW6P1
/home/tuxd3v/Desktop/H6/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
/home/tuxd3v/Desktop/H6/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
dirname: missing operand
Try 'dirname --help' for more information.
  GEN     include/autoconf.mk.dep
/home/tuxd3v/Desktop/H6/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
  GEN     include/autoconf.mk
/home/tuxd3v/Desktop/H6/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  CHK     include/generated/version_autogenerated.h
/home/tuxd3v/Desktop/H6/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
  CC      lib/asm-offsets.s
  CC      arch/arm/lib/asm-offsets.s
/home/tuxd3v/Desktop/H6/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
/home/tuxd3v/Desktop/H6/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
/home/tuxd3v/Desktop/H6/OrangePiH6/uboot/./Kbuild:34: recipe for target 'lib/asm-offsets.s' failed
make[1]: *** [lib/asm-offsets.s] Error 127
make[1]: *** Waiting for unfinished jobs....
/home/tuxd3v/Desktop/H6/OrangePiH6/uboot/./Kbuild:83: recipe for target 'arch/arm/lib/asm-offsets.s' failed
make[1]: *** [arch/arm/lib/asm-offsets.s] Error 127
Makefile:1134: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
"

I noticein **scripts/uboot_compile.sh**
TOOLS=$ROOT/toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi**hf-**

But this doesn't exist...what exist is:
toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-

Any clue?

Previously I was able to compile all the stuff with 3.10 kernel..but deleted the folder and later cloned again..it doesn't work now..

leeboby, 2019-03-08:
Please install lib32stdc++6

$ sudo apt-get install lib32stdc++6

closed 2019-03-08 by orangepi-xunlong

tuxd3v, 2019-03-08:
Woow,
omg..the troubles I had with this..
Thanks in Advance, for your reply,
It was indeed the problem...

Thanyou one more time!
Best Regards,
tux