orangepi-xunlong/OrangePi_Build #54: Output Image Doesn't Boot OrangePi 3

actsasflinn, 2019-03-10:
The image produced does not fully boot to a login prompt. Initial boot seems fast then sometimes it hangs a bit before `crng init done` but doesn't seem to go further.

I'm building for OrangePi H6 Linux 4.9 successfully on a clean vm using Ubuntu 18.04 Bionic or Ubuntu 18.10. Build, and image creation is successful.

```
sudo apt-get update
sudo apt-get upgrade
sudo reboot

sudo apt-get -y install git unzip
git clone https://github.com/orangepi-xunlong/OrangePi_Build
cd OrangePi_Build
./Build_OrangePi.sh
```

I enter my root password
Select "orangepi 3"
Select "linux4.9"

```
cd ../OrangePiH6_Linux4.9/
```

I then make a few changes to scripts/00_rootfs_build.sh to deal with an error I was encountering due to locale-gen not being installed.

```
diff --git a/00_rootfs_build.sh b/00_rootfs_build.sh
index 7e68c77..21a4abf 100755
--- a/00_rootfs_build.sh
+++ b/00_rootfs_build.sh
@@ -336,8 +336,12 @@ case $DISTRO in
                cat > "$DEST/second-phase" <<EOF
 #!/bin/sh
 export DEBIAN_FRONTEND=noninteractive
+apt-get -y update
+apt-get -y install locales
+echo en_US UTF-8 >> /etc/locale.gen
 locale-gen en_US.UTF-8
 apt-get -y update
+apt-get -y update
 apt-get -y install dosfstools curl xz-utils iw rfkill wpasupplicant usbutils alsa-utils $EXTRADEBS
 apt-get -y install rsync u-boot-tools vim parted network-manager usbmount git autoconf gcc libtool libsysfs-dev pkg-config libdrm-dev xutils-dev hostapd dnsmasq
 apt-get -y remove --purge ureadahead
```

```
./build.sh
```

Select "OrangePI 3"
I enter my root password
Select "Build Release Image"
Select "Ubuntu Xenial"
Select "Server"

I then scp the image file to my local machine OrangePiH6_Linux4.9/output/images/OrangePi_3_xenial_server_v1.0.img.tar.gz
`tar -xzvf OrangePi_3_xenial_server_v1.0.img.tar.gz`

I then use belenaEtcher to flash my 32GB Samsung microSD card. All goes well and I insert it into my Orange Pi 3 which is connected via HDMI to my TV.

Boot is fast initially then slows down and never gets to login prompt

actsasflinn, 2019-03-10:
It's worth noting there is some distortion of the text near the bottom of the screen. If I plug in a USB device it changes it sometime and clears the distortion.

orangepi-xunlong, 2019-12-06:
Now that you're ready to use the latest code, go back to OrangePi_Build and download OrangePi 3's Linux SDK. Supported build environment is Ubuntu Bionic 18.04 x64. Other environments are not yet tested.

closed 2019-12-06 by orangepi-xunlong