orangepi-xunlong/OrangePi_Build #55: Problem After build Orangepi_Oneplus kernel 4.9

tuxd3v, 2019-03-21:
Hello all,
After  I build up a image, via deboostrap, I noticed that the CPUs are there:

`root@H6:~# cat /proc/cpuinfo 
processor	: 0
BogoMIPS	: 48.00
Features	: fp asimd aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 48.00
Features	: fp asimd aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 48.00
Features	: fp asimd aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 48.00
Features	: fp asimd aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4`

But, I only have one active :(
`root@H6:~# nproc
1`

This is my uname line:
`root@H6:~# uname -a
Linux H6 4.9.118+ #2 SMP PREEMPT Mon Mar 18 01:51:04 WET 2019 aarch64 GNU/Linux`

PAssing Binary generated file to source:
`root@H6:/boot/orangepi# dtc -I dtb -O dts -o OrangePiH6.dtb OrangePiH6.dts
FATAL ERROR: Couldn't open "OrangePiH6.dts": No such file or directory
root@H6:/boot/orangepi# dtc -I dtb -O dts -o OrangePiH6.dts OrangePiH6.dtb
Warning (unit_address_vs_reg): Node /clocks has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /soc@03000000 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@07022000/s_twi0@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@07022000/s_cir0@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@07022000/twi_para@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@07022000/pwm16@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@07022000/pwm16@1 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@07022000/s_uart0@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@07022000/s_rsb0@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@07022000/s_jtag0@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@0300b000/twi3@1 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@0300b000/ts0@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@0300b000/ts0_sleep@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@0300b000/ts1@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@0300b000/ts1_sleep@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@0300b000/ts2@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@0300b000/ts2_sleep@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pinctrl@0300b000/ts3@0 has a unit name, but no reg property
[... 112 lines cut ...]
Warning (unit_address_vs_reg): Node /soc@03000000/pmu@0/powerkey@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pmu@0/regulator@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pmu@0/axp_gpio@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/pmu@0/charger@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /soc@03000000/thermal_sensor has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /soc@03000000/gpu_cooling has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /soc@03000000/keyboard has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /n_brom has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /s_brom has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /sram_ctrl has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /sram_a1 has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /sram_a2 has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /prcm has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /s_cpuscfg has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /ion/heap_sys_user@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /ion/heap_sys_contig@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /ion/heap_cma@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /ion/heap_secure@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /dramfreq has a reg or ranges property, but no unit name`

Lots of warnings..
Bellow my Device Tree Source File,
I hadn't touched its configs:

[OrangePiH6.dts.txt](https://github.com/orangepi-xunlong/OrangePi_Build/files/2992930/OrangePiH6.dts.txt)

tuxd3v, 2019-03-21:
Updating,
Adding dmesg Output..

For what I see it kills CPU 1,2,3

Can any one help?
[dmesg.txt](https://github.com/orangepi-xunlong/OrangePi_Build/files/2992999/dmesg.txt)

Thanks in Advance,
Best Regards

orangepi-xunlong, 2019-12-06:
It may be helpful to turn off these kernel configurations.  You can try the latest H6 Linux SDK.
```
-CONFIG_CPU_AUTOHOTPLUG=y
-CONFIG_CPU_AUTOHOTPLUG_ROOMAGE=y
-CONFIG_CPU_AUTOHOTPLUG_STATS=y
-CONFIG_CPU_AUTOHOTPLUG_INPUT_EVNT_NOTIFY=y
+# CONFIG_CPU_AUTOHOTPLUG is not set
```

closed 2019-12-06 by orangepi-xunlong

tuxd3v, 2020-04-08:
hello orangepi-xunlong,
yeah that was the magic bullet :)

Thanks a lot for remind me about that :)

Best Regards,