orangepi-xunlong/OrangePi_Build #56: Orange Pi 2G build with kernel 5.0

boomfly, 2019-03-24:
Hello! Is it possible to build a system on kernel 5.0. Added RDA support to the main linux repository https://github.com/torvalds/linux/tree/master/arch/arm/mach-rda

Is it possible to use their kernel to build a system for OrangePi 2G?

txurtxil, 2019-03-25:
what are the advantages?I have patched the 3.10.62 so that it does not rotate the MAC, also we have patches to operate TFT touch screen and oled in said version of the kernel, for my part I do not see any sense of uploading the kernel version.

Thore-Krug, 2019-06-12:
Why ?  Newer Kernels are safer. Also 3.10 is not supported anymore.. U better run your Hardware with a recent Kernel.

rradar, 2019-09-04:
Please only make use of supported kernels.

Drop all support for old/unsupported kernels like the 3.10 @txurtxil is still using!

Thore-Krug, 2019-12-16:
@boomflyl @txurtxil  U can use the armv7_multi_defconfig in the upstream kernel but u need to enable ARCH_RDA in the menuconfig. For Uboot no upstream version is available yet.