orangepi-xunlong/OrangePi_Build #59: [OrangePi 3] Compiling U-boot returns error: arm-linux-gnueabi-gcc: Syntax error: "(" unexpected

fbaldassarri, 2019-05-30:
Trying to build my own image for the OrangePi 3. Both with Kernel 3.10 or 4.9 and both Ubuntu or Debian, during the Compile U-boot phase it returns:

> Compile U-boot......
> Configuring for sun50iw6p1 - Board: sun50iw6p1, Options: SUN50IW6P1
> /home/fbald_000/orangepi/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/orangepi/OrangePiH6/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected

Then, the process crashes.

fbaldassarri, 2019-05-31:
Full log:

`Create uEnv.txt
 Build initrd.img
/home/fbald_000/OrangePiH6_Linux4.9
Compile U-boot......
Configuring for sun50iw6p1 - Board: sun50iw6p1, Options: SUN50IW6P1
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
dirname: missing operand
Try 'dirname --help' for more information.
  GEN     include/autoconf.mk.dep
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
  GEN     include/autoconf.mk
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  HOSTCC  scripts/basic/fixdep
  UPD     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-ld.bfd: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-ld.bfd: Syntax error: "(" unexpected
  UPD     include/generated/version_autogenerated.h
  CC      lib/asm-offsets.s
  CC      arch/arm/lib/asm-offsets.s
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
/home/fbald_000/OrangePiH6_Linux4.9/uboot/./Kbuild:34: recipe for target 'lib/asm-offsets.s' failed
make[1]: *** [lib/asm-offsets.s] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
/home/fbald_000/OrangePiH6_Linux4.9/uboot/./Kbuild:83: recipe for target 'arch/arm/lib/asm-offsets.s' failed
make[1]: *** [arch/arm/lib/asm-offsets.s] Error 2
Makefile:1134: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2`

fbaldassarri, 2019-05-31:
The full log:

 Create uEnv.txt
 Build initrd.img
/home/fbald_000/OrangePiH6_Linux4.9
Compile U-boot......
Configuring for sun50iw6p1 - Board: sun50iw6p1, Options: SUN50IW6P1
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
dirname: missing operand
Try 'dirname --help' for more information.
  GEN     include/autoconf.mk.dep
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
  GEN     include/autoconf.mk
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  HOSTCC  scripts/basic/fixdep
  UPD     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-ld.bfd: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-ld.bfd: Syntax error: "(" unexpected
  UPD     include/generated/version_autogenerated.h
  CC      lib/asm-offsets.s
  CC      arch/arm/lib/asm-offsets.s
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
/home/fbald_000/OrangePiH6_Linux4.9/uboot/./Kbuild:34: recipe for target 'lib/asm-offsets.s' failed
make[1]: *** [lib/asm-offsets.s] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: 1: /home/fbald_000/OrangePiH6_Linux4.9/uboot/../toolchain/gcc-linaro-aarch/gcc-linaro/bin/arm-linux-gnueabi-gcc: Syntax error: "(" unexpected
/home/fbald_000/OrangePiH6_Linux4.9/uboot/./Kbuild:83: recipe for target 'arch/arm/lib/asm-offsets.s' failed
make[1]: *** [arch/arm/lib/asm-offsets.s] Error 2
Makefile:1134: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2

orangepi-xunlong, 2019-12-06:
Now that you're ready to use the latest code, go back to OrangePi_Build and download OrangePi 3's Linux SDK. Supported build environment is Ubuntu Bionic 18.04 x64. Other environments are not yet tested.

closed 2019-12-06 by orangepi-xunlong