orangepi-xunlong/OrangePi_Build #61: Build ERROR: Run `./create_image` failed

YunusWang, 2019-07-12:
The error is as follows:

Creating root filesistem in local directory linux-xenial...

DEBOOTSTRAP, FIRST STAGE
OK.                        
Creating "sources.list"
Error while loading /bin/bash: No such file or directory
================================================
ERROR on second-stage, instalation NOT FINISHED.
================================================

Is there a solution?

HuangXinf, 2019-07-18:
because no debootstrap command in create_image, so wget command no execute.
you try:
sudo apt-get install debootstrap

orangepi-xunlong, 2019-12-06:
Now that you're ready to use the latest code, go back to OrangePi_Build and download OrangePi Zero's Linux SDK.

closed 2019-12-06 by orangepi-xunlong