orangepi-xunlong/OrangePi_Build #63: android7-h3-bsp

rradar, 2019-09-04:
Hello,

we are badly searching android7-h3-bsp.

Please provide links.

Thank's

rradar, 2019-09-06:
Basically a source code drop is needed. Please provide. Thank's

orangepi-xunlong, 2019-12-06:
You can go [www.orangepi.org](http://www.orangepi.org/downloadresources/) to download, android7-h3-bsp has been uploaded to Google network drive.
 

[image]

[image]

closed 2019-12-06 by orangepi-xunlong

joek85, 2020-04-01:
hello, if you want Android 7.0 source check my repo here
https://github.com/joek85/Android-7.0
it is based on android-7.0.0_r36, if you want to download the image file go to releases. it is made for OPI PLUS 2E, let me know your thoughts.
it includes GAPPS but it is a bit slow, without GAPPS it runs very smoothly.
currently i put it on hold but if you liked it, maybe we can help each other improving it, thank you.