orangepi-xunlong/OrangePi_Build #65: The script is asking for root pw... but refuses it..

krrkrr, 2019-12-17:
Hi...

I'm a basic Debian Server... I've created a basic user to avoid running the scripts as root.

When I launch the scripts, it asks me the root password. But it keeps telling me it's not the correct one... and of course, I've checked dozen of times... including trying to do it in QWERTY/AZERTY/QWERTZ just in case.. 

what can I do ?

krrkrr, 2019-12-17:
tried on several VMs ... with different flavors of Debian. Same.

orangepi-xunlong, 2020-02-26:
You first need to ensure that the current user has root privileges.

If you're testing H2+/H3/H6 series development board, use it on ubuntu18.04 desktop, which is where we tested it. Thanks for your feedback!

closed 2020-02-26 by orangepi-xunlong