orangepi-xunlong/OrangePi_Build #66: Failed to build on H3 mainline kernel.

alex-theoto, 2020-02-02:
On Debian testing (11) after fixing repository's broken url and set to install the Debian Buster, I got the following error:
`I: Running command: chroot rootfs /debootstrap/debootstrap --second-stage`
`chroot: failed to run command ‘/debootstrap/debootstrap’: Exec format error`
Where is the script for this command? I cannot find with grep.

alex-theoto, 2020-02-08:
Never mind, somehow this problem solved.

closed 2020-02-08 by alex-theoto

78bash, 2020-02-22:
Hello, any idea how this was solved?

alex-theoto, 2020-02-22:
The problem solved a few days later. I don't know what was wrong, and since I used Debian testing, probably there was something wrong with my compiling programs.
For the time being, problem solved.