orangepi-xunlong/OrangePi_Build #68: Debian mainline build fail (wget: unable to resolve host address ‘ftp2.debian.org’)

78bash, 2020-02-21:
Building Debian Stretch server with mainline kernel returned:
`...`
`CC      spl/lib/panic.o`
`CC      spl/lib/tiny-printf.o`
`LD      spl/drivers/serial/built-in.o`
`LD      spl/drivers/built-in.o`
`CC      spl/lib/strto.o`
`LD      spl/lib/built-in.o`
`LD      spl/u-boot-spl`
`OBJCOPY spl/u-boot-spl-nodtb.bin`
`COPY    spl/u-boot-spl.bin`
`MKSUNXI spl/sunxi-spl.bin`
`BINMAN  u-boot-sunxi-with-spl.bin`
`CFGCHK  u-boot.cfg`
`Complete U-boot compile....`
`Destination /home/trenaz/OrangePiH2_mainline/output/rootfs not found or not a directory.`
`Create /home/trenaz/OrangePiH2_mainline/output/rootfs`
`/tmp/tmp.KG4uUCFEVx`
`--2020-02-21 23:08:33--  http://ftp2.debian.org/debian/pool/main/d/debian-archive-keyring/debian-archive-keyring_2019.1_all.deb`
`Resolving ftp2.debian.org (ftp2.debian.org)... failed: Name or service not known.`
`wget: unable to resolve host address ‘ftp2.debian.org’`

Pinging returns:
`ping: ftp2.debian.org: Name or service not known`

`PING google.com (172.217.19.46) 56(84) bytes of data.`
`64 bytes from ham02s11-in-f46.1e100.net (172.217.19.46): icmp_seq=1 ttl=49 time=67.5 ms`
`64 bytes from ham02s11-in-f46.1e100.net (172.217.19.46): icmp_seq=2 ttl=49 time=67.4 ms`
`64 bytes from ham02s11-in-f46.1e100.net (172.217.19.46): icmp_seq=3 ttl=49 time=67.2 ms`

alex-theoto, 2020-02-22:
Script is outdated.
I made a pull request with some modifications that I made to install Debian Buster.
Although I'm not familiar with github, the updated scripts are at my repository and waiting for acceptance from the original author.

orangepi-xunlong, 2020-02-26:
Sorry, this should be changed to **ftp.debian.org**.

How about the CDN option? This should work.
[image]

closed 2020-02-26 by orangepi-xunlong