orangepi-xunlong/OrangePi_Build #69: git clone  RPC failed

zhouyinwei, 2020-03-01:
克隆Orange Pi 时，提示RPC failed，修改了git config的http.postBuffer依然无效，应该是服务器设置问题

orangepi-xunlong, 2020-03-02:
Hi，

能提供下详细的信息吗，我们在 ubuntu 下测试还没发现有这个问题，下载代码如果速度很慢，或者下到中途就断了，一般是网络限制的原因。

zhouyinwei, 2020-03-02:
Git已经配置成如下设置了：
utech@ubuntu:~/OrangePi_Build$ git config -l
http.postbuffer=2048576000
core.compression=-1
core.repositoryformatversion=0
core.filemode=true
core.bare=false
core.logallrefupdates=true
remote.origin.url=git@github.com:orangepi-xunlong/OrangePi_Build.git
remote.origin.fetch=+refs/heads/*:refs/remotes/origin/*
branch.master.remote=origin
branch.master.merge=refs/heads/master

zhouyinwei, 2020-03-02:
utech@ubuntu:~/OrangePi_Build$ ./Build_OrangePi.sh 
 Download from : https://github.com/orangepi-xunlong/OrangePiRK3399_kernel.git 
Cloning into '/home/utech/OrangePiRK3399_Pi4/kernel'...

remote: Enumerating objects: 64295, done.
remote: Counting objects: 100% (64295/64295), done.
remote: Compressing objects: 100% (52321/52321), done.
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

orangepi-xunlong, 2020-03-02:
Hi,

这个是由于网络限制导致的，和服务器配置没有关系，建议通过码云中转的方式来下载。

zhouyinwei, 2020-03-02:
> Hi,
> 
> 这个是由于网络限制导致的，和服务器配置没有关系，建议通过码云中转的方式来下载。

好的，我试试中转到码云再下载

orangepi-xunlong, 2020-03-02:
保存完对应SDK的几个仓库到码云后，可以在 RK3399SDK_BuildEnvironment.sh 脚本中修改下 git 服务器的地址为码云的地址，然后再下载
[image]

zhouyinwei, 2020-03-02:
> 保存完对应SDK的几个仓库到码云后，可以在 RK3399SDK_BuildEnvironment.sh 脚本中修改下 git 服务器的地址为码云的地址，然后再下载
> [image]

OK，thanks

closed 2020-03-03 by orangepi-xunlong