orangepi-xunlong/OrangePi_Build #70: 使用Ubuntu 16.04版本编译OrangePi 4时报如下错误卡住不动了

zhouyinwei, 2020-03-02:
Setting up x11proto-xf86dri-dev (2018.4-4) ...
Setting up libxinerama-dev:arm64 (2:1.1.3-1) ...
Setting up libgbm-dev:arm64 (19.2.8-0ubuntu0~18.04.2) ...
Setting up x11proto-gl-dev (2018.4-4) ...
Setting up libxv-dev:arm64 (2:1.0.11-1) ...
Setting up libcapnp-0.6.1:arm64 (0.6.1-1ubuntu1) ...
Setting up x11proto-resource-dev (2018.4-4) ...
Setting up libxcb-xkb-dev:arm64 (1.13-2~ubuntu18.04) ...
Setting up x11proto-xf86dga-dev (2018.4-4) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
qemu: Unsupported syscall: 277

baiywt, 2020-03-03:
卡了多久？可能跟机器性能也有关系，等待一段时间。另外最好使用ubuntu18.04桌面版进行编译。

zhouyinwei, 2020-03-03:
> 卡了多久？可能跟机器性能也有关系，等待一段时间。另外最好使用ubuntu18.04桌面版进行编译。

使用18.04版本编译过了，应该是版本兼容性的问题

closed 2020-03-03 by zhouyinwei