orangepi-xunlong/OrangePi_Build #73: Orangepi_4使用GPU跑opencl的算法，跑一段时间后，算法程序就会卡住

Lemonning, 2020-03-26:
基于Mali-t860跑opencl算法，依赖libmali-midgard-t86x-r14p0.so动态库，算法程序循环执行，执行几次（不固定，几次到几十次不等）之后程序会卡住；该程序在其他基于的RK3399平台的开发板运行过，没有出现问题，请分析下原因。
是不是kernel版本或者GPU相关驱动版本太低了，没有合入RK最新的补丁？

orangepi-xunlong, 2020-03-29:
我们要检查一下------------------ 原始邮件 ------------------

 

 
&mdash;

Lemonning, 2020-04-20:
[TestDemo.tar.gz](https://github.com/orangepi-xunlong/OrangePi_Build/files/4501482/TestDemo.tar.gz)
1、附件是我的demo，使用opencl对图片循环做高斯滤波。解压后运行 “./imgGaussBin”即可（需要的动态库已放到tar包），问题是循环一定次数之后程序会卡住；
2、有一块其他厂家的RK3399方案的开发板，也有这个问题，他们反馈驱动和mali库版本不匹配；替换了新的动态库“libmali_rk3399_x11_gbm_arm64_r18p0.so”，并在内核编译时关闭了选项“CONFIG_MALI_MIDGARD_FOR_LINUX=y”；问题得到解决。
我在4B的源码中关闭该选项后，编译默认就选择了“CONFIG_MALI_MIDGARD_FOR_ANDROID”。无法在linux系统中加载gpu。
请帮忙分析下。