orangepi-xunlong/OrangePi_Build #80: 官方源码编译不过

minchieh-fay, 2020-04-06:
编译环境：ubuntn:18.04
drivers/net/wireless/rockchip_wlan/rtl8822be/hal/hal_btcoex.c: In function 'hal_btcoex_IsBtControlLps':
drivers/net/wireless/rockchip_wlan/rtl8822be/hal/hal_btcoex.c:3890:2: error: invalid use of undefined type 'struct btc_coexist'
  if (GLBtCoexist.bdontenterLPS == _TRUE)
  ^~
drivers/net/wireless/rockchip_wlan/rtl8822be/hal/hal_btcoex.c:3896:2: error: invalid use of undefined type 'struct btc_coexist'
  if (GLBtCoexist.bt_info.bt_disabled)
  ^~
drivers/net/wireless/rockchip_wlan/rtl8822be/hal/hal_btcoex.c:3899:2: error: invalid use of undefined type 'struct btc_coexist'
  if (GLBtCoexist.bt_info.bt_ctrl_lps)
  ^~

baiywt, 2020-04-07:
检查下kernel和scripts是否是最新的，toolchain的目录结构如下：
csy@ubuntu:toolchain$ tree -L 1
.
├── gcc-linaro-6.3.1-2017.05-x86_64_aarch64-linux-gnu
└── README.md

1 directory, 1 file

另外源码有没有自己做过改动？

minchieh-fay, 2020-04-07:
怕有问题，重新下载的最新的，没改过任何代码

minchieh-fay, 2020-04-07:
工具链也没问题，内核都编好了，卡网卡驱动上了

baiywt, 2020-04-07:
这个在ubuntu-18.04.1-desktop-amd64上测试过，没有遇到过这种错误。请检查一下kernel有没有缺失文件，这些类型在.h中是有定义的，很可能是由于文件缺失导致的。docker镜像我们需要测试一下，后面可以在手册上写个说明。

minchieh-fay, 2020-04-07:
好的，我用的是Ubuntu 18.04的docker

minchieh-fay, 2020-04-09:
18.04.1 desktop确实可以

closed 2020-04-10 by orangepi-xunlong