orangepi-xunlong/OrangePi_Build #84: compile rootfs error: chroot: failed to run command ‘mount’: Exec format error

chadsea, 2020-06-03:
我的编译环境是： ubuntu 18.04 x64
编译rootfs时报错如下：
    Extracting ... OK
    chroot: failed to run command ‘mount’: Exec format error
    chroot: failed to run command ‘mount’: Exec format error
    chroot: failed to run command ‘/second-phase’: Exec format error
这个错误产生在以下位置：
  文件路径：scripts/lib/distributions.sh
    do_chroot() {
        # Add qemu emulation.
        cp /usr/bin/qemu-arm-static "$DEST/usr/bin"
        cmd="$@"
        chroot "$DEST" mount -t proc proc /proc || true
        chroot "$DEST" mount -t sysfs sys /sys || true
        chroot "$DEST" $cmd
        chroot "$DEST" umount /sys
        chroot "$DEST" umount /proc

        # Clean up
        rm -f "$DEST/usr/bin/qemu-arm-static"
    }
能告诉我怎么解决这个问题吗？是qemu的问题吗

orangepi-xunlong, 2020-06-04:
Try reinstall qemu-user-static

sudo apt purge qemu-user-static
sudo apt install qemu-user-static

chadsea, 2020-06-05:
Thank you very much, it works now

closed 2020-06-05 by chadsea