orangepi-xunlong/OrangePi_Build #89: No build

techaura, 2020-08-07:
I: Running command: chroot rootfs /debootstrap/debootstrap --second-stage
chroot: failed to run command ‘/debootstrap/debootstrap’: Exec format error

techaura, 2020-08-07:
Orange PI 3
Debian Stretch
Install host tools
 Build U-boot 
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
scripts/kconfig/conf  --syncconfig Kconfig
  CHK     include/config.h
  UPD     include/config.h
  CFG     u-boot.cfg
  GEN     include/autoconf.mk.dep
  CFG     spl/u-boot.cfg
  GEN     include/autoconf.mk
  GEN     spl/include/autoconf.mk
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/dtc.o
  SHIPPED scripts/dtc/pylibfdt/libfdt.i
  PYMOD   scripts/dtc/pylibfdt/_libfdt.so
  UPD     include/config/uboot.release
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
./"board/sunxi/mksunxi_fit_atf.sh" \
arch/arm/dts/sun50i-h6-orangepi-3.dtb > u-boot.its
  LEX     scripts/dtc/dtc-lexer.lex.c
  YACC    scripts/dtc/dtc-parser.tab.h
  YACC    scripts/dtc/dtc-parser.tab.c
  CHK     include/generated/version_autogenerated.h
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  UPD     include/generated/version_autogenerated.h
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  CC      lib/asm-offsets.s
  CC      arch/arm/lib/asm-offsets.s
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
  UPD     include/generated/asm-offsets.h
  UPD     include/generated/generic-asm-offsets.h
  LDS     u-boot.lds
  HOSTLD  scripts/dtc/dtc
  HOSTCC  tools/gen_eth_addr
  HOSTCC  tools/gen_ethaddr_crc.o
  HOSTCC  tools/img2srec
  WRAP    tools/lib/crc8.c
  ...
I: Extracting dash...
I: Extracting libdb5.3...
I: Extracting debconf...
I: Extracting debianutils...
I: Extracting diffutils...
I: Extracting dpkg...
I: Extracting e2fslibs...
I: Extracting e2fsprogs...
I: Extracting libcomerr2...
I: Extracting libss2...
I: Extracting findutils...
I: Extracting gcc-6-base...
I: Extracting libgcc1...
I: Extracting libc-bin...
I: Extracting libc6...
I: Extracting multiarch-support...
I: Extracting grep...
I: Extracting gzip...
I: Extracting hostname...
I: Extracting init-system-helpers...
I: Extracting libcap-ng0...
I: Extracting libgcrypt20...
I: Extracting libgpg-error0...
I: Extracting libselinux1...
I: Extracting libsemanage-common...
I: Extracting libsemanage1...
I: Extracting libsepol1...
I: Extracting lsb-base...
I: Extracting liblz4-1...
I: Extracting mawk...
I: Extracting libncursesw5...
I: Extracting libtinfo5...
I: Extracting ncurses-base...
I: Extracting ncurses-bin...
I: Extracting libpam-modules...
I: Extracting libpam-modules-bin...
I: Extracting libpam-runtime...
I: Extracting libpam0g...
I: Extracting libpcre3...
I: Extracting perl-base...
I: Extracting sed...
I: Extracting sensible-utils...
I: Extracting login...
I: Extracting passwd...
I: Extracting libsystemd0...
I: Extracting libudev1...
I: Extracting sysvinit-utils...
I: Extracting tar...
I: Extracting tzdata...
I: Extracting libustr-1.0-1...
I: Extracting bsdutils...
I: Extracting libblkid1...
I: Extracting libfdisk1...
I: Extracting libmount1...
I: Extracting libsmartcols1...
I: Extracting libuuid1...
I: Extracting mount...
I: Extracting util-linux...
I: Extracting liblzma5...
I: Extracting zlib1g...
I: Running command: chroot rootfs /debootstrap/debootstrap --second-stage
chroot: failed to run command ‘/debootstrap/debootstrap’: Exec format error
adminus@ubuntu18:~/OrangePiH6_mainline$ 

.img not created

leeboby, 2020-08-11:
Try reinstall qemu-user-static

sudo apt purge qemu-user-static
sudo apt install qemu-user-static

techaura, 2020-08-18:
Try reinstall qemu-user-static

sudo apt purge qemu-user-static
sudo apt install qemu-user-static
leeboby(C)
done

It would be nice to point in the manual all the necessary programs that need to be installed in order for the assembly to assemble.
There's a really big list there.

closed 2020-08-18 by techaura