orangepi-xunlong/OrangePi_Build #90: What to do after built image

vslzl, 2020-08-12:
The documentation lacks of the information about flashing the image to the device. We've finished all steps successfully and at final step we selected the Build release image. But the output was some files. So what to do next? how we flash the output to the card? Any help will be appreciated.

Sergio-, 2020-08-16:
You can flash the .img file to the SD Card and boot your device from it.

techaura, 2020-08-19:
The specific help will depend on which system you want to record to the SD card.
The file with the name of the Linux version and the .img extension must be written to the microSD card with a suitable program.
For example, my file named: OrangePi_3_debian_stretch_server_linux5.3.5_v2.0.2.img.
You have a file with a similar name as the version of what you have built.
I record under Windows so I use Balena Etcher, you can record with another program.