orangepi-xunlong/OrangePi_Build #92: menuconfig has no effect for Orange Pi 3 H6

techaura, 2020-08-18:
Orange Pi 3 H6
Existing drivers cannot be added to the .img. I cannot add USB-Serial and other drivers that are not included in the kernel by default.
Using make menuconfig - Not helpful. There are no modules either in the kernel or in the form of  *.ko files

Sergio-, 2020-08-19:
It seems this area is abandoned, nobody seems to be looking at the issues posted here...

techaura, 2020-08-20:
.config, modules.order and other files are replaced during assembly with default ones.
The work of make menuconfig is visible only before the build, then everything is reset.
Does the user have to guess where this is happening, or is he doomed to necessarily study all the scripts to find this place with an error?