orangepi-xunlong/OrangePi_Build #103: CreatingABootableSD_Card - OrangePi_4G_IOT

kerzol81, 2020-12-28:
Hi,

I followed the readme and tried a couple of times in a virtualised Ubuntu 18.04 environment (fully upgraded), however I always get this error:

============================================
make: Entering directory '/home/zoli/OrangePi4G-iot/uboot/lk'
make[1]: Entering directory '/home/zoli/OrangePi4G-iot/uboot/lk'
libshowlogo new path ------- platform/mt6735/../../lib/libshowlogo
BOOT_LOGO = hd720
lk/logo/dir=dev/logo,builddir=/home/zoli/OrangePi4G-iot/output/lk/build-bd6737m_35g_b_m0
including app/mt_boot dev/keys dev/lcm dev/video lib/debug lib/heap lib/libc lib/libfdt lib/libshowlogo lib/ptable lib/zlib
generating /home/zoli/OrangePi4G-iot/output/lk/build-bd6737m_35g_b_m0/config.h
Compiling_BMP_TO_RAW dev/logo/hd720/hd720_uboot.bmp
dev/logo/tool/bmp_to_raw /home/zoli/OrangePi4G-iot/output/lk/build-bd6737m_35g_b_m0/dev/logo/hd720/hd720_uboot.raw dev/logo/hd720/hd720_uboot.bmp
dev/logo/tool/bmp_to_raw: error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory
dev/logo/rules.mk:180: recipe for target '/home/zoli/OrangePi4G-iot/output/lk/build-bd6737m_35g_b_m0/dev/logo/hd720/hd720_uboot.raw' failed
make[1]: *** [/home/zoli/OrangePi4G-iot/output/lk/build-bd6737m_35g_b_m0/dev/logo/hd720/hd720_uboot.raw] Error 127
make[1]: Leaving directory '/home/zoli/OrangePi4G-iot/uboot/lk'
makefile:20: recipe for target 'make-make' failed
make: *** [make-make] Error 2
make: Leaving directory '/home/zoli/OrangePi4G-iot/uboot/lk'

Do you guys succeded to create a Bootable SD card

kerzol81, 2020-12-28:
I'm experimenting with the build order now:
Build Linux -> Build kernel -> Build uboot and the release image

I hope it's gonna solve the problem, I waited 42 days to get this OrangePi4G-IOT :-)

orangepi-xunlong, 2021-01-01:
We will check it.

发自我的iPhone

 

 Build Linux -&gt; Build kernel -&gt; Build uboot and the release image

pmunozsigma, 2021-01-28:
Could you have any progress with this problem?
I have the same error and I don't know how to fix it.

kerzol81, 2021-01-28:
no. It doesnt work.