orangepi-xunlong/OrangePi_Build #105: i96 console freezes

mrx23dot, 2021-04-13:
I bought an Orange Pi i96
 - installed the official debian image,
- logged in
- queried the free disk space:
- I have  **2.9GB** free disk space:
Filesystem      Size  Used Avail Use% Mounted on
udev             10M     0   10M   0% /dev
tmpfs            46M  1.9M   44M   5% /run
/dev/mmcblk0p2  3.7G  693M  2.9G  20% /
tmpfs           115M     0  115M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           115M     0  115M   0% /sys/fs/cgroup
tmpfs           115M     0  115M   0% /tmp
/dev/mmcblk0p1   49M   12M   34M  27% /media/boot
tmpfs            23M     0   23M   0% /run/user/0

 - and it got stuck after it,  I cannot enter anything into console. 
 - even after removing power and serial cable
 - I have to re-burn the SD card, I think **something destructive happens with the config file**

 Please provide technical support!  So far it is useless.

I tried both official releases (debian + ubuntu), but same issue:
- http://www.orangepi.org/downloadresources/orangepii96/orangepii96_94bf0378a51e1277ff416638.html
- http://www.orangepi.org/downloadresources/orangepii96/orangepii96_72069b4e8e756bcc44ae393e.html
one of them: OrangePi_i96_debian_stretch_server_linux3.10.62_v0.0.4

mrx23dot, 2021-04-13:
Or here is an easy to reproduce:

- fresh debian
- login root
- orangepi-config connect to wifi
- reboot now
- on reboot it only goes this far, **every time**

[    6.351501] <rda_msys>: end of pending msg
[    6.360961] <rda_msys>: end of pending msg
[  OK  ] Mounted /media/boot.
[  OK  ] Reached target Local File Systems.
         Starting Create Volatile Files and Directories...
[  OK  ] Started Apply Kernel Variables.
[    6.590026] enable vbus and disable charge
         Starting Raise network interfaces...
[    6.860046] rda_msys: rda-codec : timeout as sending a message
[    6.860595] >>>> [aud_StreamStart], ret [1]
[  OK  ] Started Create Volatile Files and Directories.
[    7.090087] rda_msys: rda-power : timeout as sending a message
[  OK  ] Listening on Load/Save RF Kill Switch Status /dev/rfkill Watch.
         Starting Network Time Synchronization...
         Starting Update UTMP about System Boot/Shutdown...
         Starting Load/Save RF Kill Switch Status...
[    7.359985] rda_msys: rda-codec : timeout as sending a message
[    7.360595] >>>> [aud_StreamRecord], ret [1]
[    7.456359] wlan_rfkill_set 0
[  OK  ] Started Update UTMP about System Boot/Shutdown.
[  OK  ] Started Network Time Synchronization.
[  OK  ] Reached target System Time Synchronized.

**then I cannot input anything!**

mrx23dot, 2021-04-14:
this seems to solve it:
apt purge -y alsa-utils

but why?

mrx23dot, 2021-04-15:
after 20hours connecting again from serial port console I get:

[76002.189086] ttyS0: 11 input overrun(s)
[76003.242187] ttyS0: 11 input overrun(s)
[76004.292358] ttyS0: 10 input overrun(s)
[76005.317016] ttyS0: 9 input overrun(s)
[76006.464233] ttyS0: 9 input overrun(s)
[76007.836242] ttyS0: 11 input overrun(s)
[76009.560180] ttyS0: 4 input overrun(s)
[76010.860229] ttyS0: 2 input overrun(s)
[76012.011352] ttyS0: 1 input overrun(s)
[76013.069824] ttyS0: 2 input overrun(s)

TheUserDead, 2023-01-03:
The problm described in alsa-mixer software. IDK now i'ts possible, but before configure wi-fi you need remove this package alsa-utils as far i remember. Or just burn https://github.com/TheRemote/Legendary-OrangePi-i96 most problems fixed here.

TheRemote, 2023-01-04:
I personally give permission for that to happen and ask for nothing in return.  It would be awesome (for everyone).